beckus/qemu_stm32 #8: Weirdness / bug in timer 1 initialization

smunaut, 2015-10-07:
Hi,

First thanks for this project, pretty happy it exists :)

I've been trying to run some code dumped from a running system and encoutering some issues. First one is linked to timers and IRQ.

First thing is definitely a bug in stm32.c there is

```
stm32_create_timer_dev(stm32_container, STM32_TIM1, 1, rcc_dev, gpio_dev, afio_dev, 0x40012C00, tim1_irqs, 5);
```

But that last '5' should be '4' because tim1_irqs is an array with 4 entries only.

Second thing is that I'm not even sure what this tries to accomplish. Because those 4 irqs, will end up calling sysbus_connect_irq multiple times, but it will just overwrite the previous value so in the end only the last value written will be "connected" AFAICT.

Cheers,
 
   Sylvain

beckus, 2015-10-10:
Hello, Thank you for the email, and I am glad you find the project useful.  Yes, those are both definitely issues.  It looks like only one interrupt is implemented, but the initialization code is still trying to pass in all four (unsuccessfully). It seems as though this would initialize the IRQ pointer to an arbitrary value (due to the "5").  I made a new branch called stm32_timer1_fix ( beckus/qemu_stm32 ) and made a commit to fix this.  It still only implements one interrupt, but it should initialize it properly.  Please, let me know if this helps.  Someone else had written most of the timer module, and I do not currently have a program to test Timer 1 to see if this really fixes it.  If you think it looks good, please let me know, and I can merge the commit into the main stm32 branch.
Thanks again and best of luck,Andre

beckus, 2015-12-12:
I merged the patch into the main stm32 branch.

closed 2015-12-12 by beckus