beckus/qemu_stm32 #10: -kernel unsopported machine type

MatrixLing, 2016-05-09:
I test your qemu-stm32 using your stm32_p103_demos project. I use the command "make qemu_test_QEMURUN" to run. But it print the following error. I don't know how to solve it. Can you help me, thanks.

ling@UBUNTU64:~/Desktop/stm32_p103_demos$ make qemu_test_QEMURUN
killall -q qemu-system-arm
make: [qemu_test_QEMURUN] error 1 (ignore)
/home/ling/Desktop/beckus-qemu_stm32-b37686f/arm-softmmu/qemu-system-arm -M stm32-p103 -kernel demos/qemu_test/main.bin
qemu-system-arm: -kernel demos/qemu_test/main.bin: Unsupported machine type
Use -machine help to list supported machines!
make: *** [qemu_test_QEMURUN] error 1

beckus, 2016-05-13:
Hello Ling,
Can you try running this command:
qemu-system-arm -M help

This will give you a list of the supported machines.  You should see "stm32-p103" in the list.  Let me know if you see it.  If not, what commands are you using to build and compile qemu-stm32?

MatrixLing, 2016-05-17:
I have solve It.  I first download sourse from the url:https://codeload.github.com/beckus/qemu_stm32/legacy.tar.gz/master and it is not has stm32-p103.

I try to use git clone https://github.com/beckus/qemu_stm32, and it has stm32-p103.

Thanks!

beckus, 2016-05-17:
I see.  Great, glad it is working now.  Thanks for the update.

closed 2016-05-17 by beckus

vinifr, 2016-07-21:
Hi beckus, a little doubt. In your qemu_stm32 we run the firmware using -kernel command, but in qemu_pebble does not work. In pebble we have to use -pflash command. I wanted to know why...