beckus/qemu_stm32 #12: stm32 branch won't compile

yodalee, 2016-05-16:
qemu depends on glib, which has some deprecated function like `g_mem_set_vtable`. These deprecated function produce error while compile.
The bugs is fixed in master branch at commit 98cf48, however the patch doesn't apply to stm32 branch yet.

beckus, 2016-05-21:
Hello,
Yes, unfortunately the stm32 branch is pretty far behind the master branch due to some incompatible changes.  I hope to get the stm32 branch updated at some point.  I see this at the end of the Change Log for Qemu 2.6 (http://wiki.qemu.org/ChangeLog/2.6):

"Known issues
Recent versions of GTK 3 will produce compilation warnings about deprecated functions in ui/gtk.c. This should not be a problem in release tarballs (where warnings do not result in compilation failures). For building from a git tree they can be worked around by passing the --disable-werror option to configure. We plan to fix the issue for the 2.7 release."

I am not sure if this is the same exact issue you refer to, but have you tried using the "--disable-werror" option?

yodalee, 2016-05-23:
`--disable-werror` can compile the project, thx.

beckus, 2016-05-23:
OK, I'm glad to hear.  I added a comment about this option in the README.

closed 2016-05-23 by beckus