beckus/qemu_stm32 #15: UART problem with stm branch

grapherd, 2016-08-07:
Hi beckus, 
I just clone qemu_stm32 `stm32` branch and, after make it and `make uart_echo_QEMURUN`, I found that serial0 didn't answer my input anyway.

But, after downloading v0.13, and redo again, uart_echo_QEMURUN works.

I'm wondering why `stm32` branch didn't work, can you give me some clue?

thanks!

p.s. using: gcc6.1, archlinux-x64

T0jan, 2016-10-17:
i can confirm this issue. a fast workaround would be nice!

oska874, 2017-12-13:
You should look at /dev/pts/ twice: first look before running `make`,second look after `make`, then you will find that there have add a file,then you can use `minicom` to read from it:
```
sudo minicom -D /dev/pts/5
```
ps: `5` is the new file in `/dev/pts`.

beckus, 2018-07-25:
I apologize for the very long delay in responding. The problem should be fixed now in commit 415f9ab43431024b9e44a32c2aa8aed9afbc1cb6 .

closed 2018-07-25 by beckus