beckus/qemu_stm32 #28: I can't install qemu_stm32 on mac os x

posutsai, 2019-04-16:
I try to follow the installation instruction. However when I `mak`e, the compiler can't find `Cocoa/Cocoa.h` header. Following is the error message.
```
  OBJC  ui/cocoa.o
ui/cocoa.m:25:9: fatal error: 'Cocoa/Cocoa.h' file not found
#import <Cocoa/Cocoa.h>
        ^~~~~~~~~~~~~~~
1 error generated.
make: *** [ui/cocoa.o] Error 1
```
I wonder if I miss some dependency to install?

beckus, 2019-04-17:
I'm sorry, but I am not familiar with the compile process on Mac OS X. 
Have you tried compiling the latest QEMU from https://github.com/qemu/qemu to see if the issue occurs there as well? This may help with troubleshooting. Note that QEMU_STM32 is based on an older version of QEMU and so sometimes there are issues with QEMU_STM32 that have since been fixed in the main QEMU product...

kousu, 2019-06-01:
@posutsai , I'm building qemu_stm32 on OS X. Did you install the OS X toolchain using `xcode-select --install`?

Here's the script I've been using to build:

```
cd qemu_stm32

ARGS=""
ARGS="$ARGS --target-list="arm-softmmu""
ARGS="$ARGS --disable-werror"
ARGS="$ARGS --enable-debug"
ARGS="$ARGS --disable-glusterfs"
ARGS="$ARGS --python=python2"

if [ -n "$PREFIX" ]; then
  ARGS="$ARGS --prefix=$PREFIX"
fi

if [ -n "$DEBUG_STM32" ]; then
  ARGS="$ARGS --extra-cflags=-DDEBUG_STM32_RCC"
  ARGS="$ARGS --extra-cflags=-DDEBUG_STM32_UART"
  ARGS="$ARGS --extra-cflags=-DDEBUG_STM32_TIMER"
fi

./configure $ARGS
make -j4
sudo make install
```

It works reliably for me. I didn't have to do any special, I don't think.

NoeelMoeskops, 2020-03-09:
maybe a bit late, but use: 
`./configure --enable-debug \
  --target-list=”arm-softmmu” \
  --python=python2 \
  --disable-cocoa`
[source](https://medium.com/@ly.lee/stm32-blue-pill-unit-testing-with-qemu-blue-pill-emulator-9d88002a68b2)

beckus, 2021-09-12:
Thank you @kousu and @NoeelMoeskops for your comments on this issue.

closed 2021-09-12 by beckus