beckus/qemu_stm32 #40: Does it also emulate the 'missing devices' in QEMU STM32 boards ?

adi-g15, 2021-12-07:
At https://qemu.readthedocs.io/en/latest/system/arm/stm32.html, QEMU lists some available machines based on the stm32 chips, but there are lots of missing devices

A naive question, but **does this project emulate the devices listed there as "Missing Devices"** ?
I will try building it and run it, though a confirmation from your side will be helpful. Though Thanks for creating a project specifically for stm32.

beckus, 2021-12-08:
Hello,
Thank you for pointing out that page, I had not seen it. I am not familiar with the stm32vldiscovery or netduino2 machines that are a part of QEMU, as I have not had a chance to try them or look at them.

This project (QEMU STM32) implements the following modules which are on the "Missing Devices" list:
DAC
DMA
GPIO
RCC
IWDG

You can see the list of devices implemented in QMEU STM32 here (with links to the source code): http://beckus.github.io/qemu_stm32/
But, be aware that the device implementations are not complete, and in some cases are very limited.

I hope you find the project useful, and best of luck!

- Andre

adi-g15, 2021-12-08:
Thanks for the quick reply.
I was looking for the gpio part, thanks for the info, Thanks a lot 😅👍.

closed 2021-12-08 by adi-g15